3d-gussner/Prusa-Firmware #38: Combining forces

thess, 2017-11-09:
Hi @3d-gussner 

It looks like you, @Sebastianv650 and I are both on the same path to cleanup and enhance the Prusa firmware and put out "community mods". Perhaps we should combine our efforts somehow and then figure out how to manage updates and releases?

Take a look at what I've started with my own private builds and some new README, BUILDING and changelog doc over at https://github.com/thess/Prusa-Firmware/tree/private-build2

3d-gussner, 2017-11-10:
Hi @thess,

your coding skills are much better than mine and if we could combine the codes and review with 3 persons the chance of getting a PR to Prusa firmware being included may be easier.

My focus at this moment are improvements and translations as
- PJR and Jeff Jordan Hyperfine Bed tuning, which is just great. Don't get it why Prusa didn't pulled one of our PRs.
- Filament runout sensors which are also coming soon with the MK3 but give the thinker the option to choose what they can get their hands on and feel comfortable with like mechanical and optical switches.

I also think having different languages available would make the use for non-english speaking people especially kids much easier.
Because of the limits of the ATmega 2560 i even cannot an additional language anymore.
Portuguese translation has been provided by @nfsmedeiros and somebody is busy with the French translation.
It maybe waste of time, but we will see if people are interested in these. 

BTW: Nice BUILDING instructions and also like the CHANGELOG. Don't know why Prusa doesn't update their build instructions as they use intern Arduino IDE 1.8.3 or 1.8.4

Was thinking of using the Github Wiki for new features.

It is a pity that Prusa Firmware is so far away from the Marlin code, as there are updates, fixes nearly every day.
Example: My https://github.com/prusa3d/Prusa-Firmware/pull/238 using M600 g-code for the filament runout sensor comes from Marlin.

So what is your idea how to get it done? A new Github repository where we agree to and publish links to that one?

Sebastianv650, 2017-11-10:
The reason why I created the "community mod" branch was that quite a lot of good PRs and even bug fixes against Prusa branch where not merged at this time, so I wanted to give users the possibility to download a "complete" code base without knowing how to create an own fork, merge the PRs and so on.

Beside maintaining the LIN_ADVANCE code (which seems to be done, but you never know..) I have no big intention to add further code at the moment, mainly because
- Prusa FW is only for the Prusa printers, while coding for Marlin reaches nearly every printer and
- I'm a little bit confused and repelled by the Prusa code, it's not a quarter as structured as Marlin - which is a big thing for a person like me who is only coding from time to time so I have not a lot of experience. It's just hard to read and understand this way.

Therefore it won't hurt myself if one of you maintain a more advanced comunity mod which we all use for PR until they get merged into the original FW. And if there is an interesting topic to code on or something to review I will be there :)

boelle, 2018-08-04:
i have had similar thoughts lately... 

ie adding support for Ramps+2130's which is very close to the mk3

so i have started the work of copying the einsy pin file and modding it  to work on a ramps board

there are a few things in the pins file i dont understand

VOLT_PWR_PIN and VOLT_BED_PIN is this just a simple resistor divider that tells when power is applied?
or is one of them a resistor devider and the other the power panic?

TACH_0 and TACH_1, are those the rmp signal straight to a digital pin?

of course the hardest thing of them all would be to get support for ful graphic displays, even just 4 or 2 text lines like on the text display would be better than to trash the display just to get another one. even thou a display is "cheap" i think most hate to waste money if the display they have is ok

link to my github repro: https://github.com/boelle/Prusa-Firmware/tree/MK3/Firmware

cimoalpacino, 2018-08-04:
PE2 or digital pin 2 is basically XMAX on RAMPS.

I'm also trying to port MK3 fw with power panic to RAMPS.
So far I've been bale to port it with some previous versions, but I'm unable to connect it with pronterface....it messes up my communication.
Still trying to figure out whats wrong (don't have a working fw on git, yet).

But I think our goal is the same :)

boelle, 2018-08-04:
yes.... but what file in the prusa firmware tells what pin is power panic? like this one which says chipselect pin for the 2130 stepper on the x axis is pin 41

#define X_TMC2130_CS           41

there must be one like it for power panic.... but in what file??

cimoalpacino, 2018-08-04:
You can find it in Marlin.h

I have modified it to the code below.
`
#if MOTHERBOARD == BOARD_RAMPS_14_EFB
	//#define UVLO !(PIND & (1<<2)) //PD2 (D19) = ZMAX
	#define UVLO !(PINE & (1<<4)) //PE4 (D2) = XMAX
#else
	#define UVLO !(PINE & (1<<4)) //PE4 (D2) = XMAX
#endif`

You can find all my changes to the FW on my git
[https://github.com/cimoalpacino/Prusa-Firmware-1/tree/MK3/Firmware](url)

boelle, 2018-08-04:
found it here: https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/Marlin.h#L466

boelle, 2018-08-04:
the line just says #define UVLO !(PINE & (1<<4))

how do we know fore sure this means power panic is digital 2 ??

cimoalpacino, 2018-08-04:
PINs, port E, left shift for 4, means pin = PE4

From this link [https://www.arduino.cc/en/uploads/Hacking/PinMap2560big.png](url) you can clearly see this is equal to digital pin D2.
regards,

thess, 2018-08-04:
Thread is old and is now OT

closed 2018-08-04 by thess

boelle, 2018-08-04:
well it does not prevent me and @cimoalpacino from writing together

not unless @3d-gussner thinks we should not

boelle, 2018-08-04:
@cimoalpacino but yes... i found a better diagram here

http://icircuit.net/wp-content/uploads/2014/09/megapdf.jpg

digital 2 is PE4

PINE  that is the same as PE  
1<<4  is the same as 4 

correct?

so for digital 40 that is PL5

PINL 
and 1<<5

or am i wrong?

cimoalpacino, 2018-08-04:
Physical pin is 40, but name of this digital pin is 44.
See the link I sent you earlier.

cimoalpacino, 2018-08-04:
#define UVLO !(PINL & (1<<5)) = PL5 (pin on chip is nr. 40, naming is Digital pin 44)

boelle, 2018-08-04:
oh... digital 40 is PG1 right?

boelle, 2018-08-04:
so #define UVLO !(PING & (1<<1))  ?

boelle, 2018-08-04:
the trick here is to find free pins on ramps board when a LCD is connected, so a lot of going back and fort between schematics and pinouts

cimoalpacino, 2018-08-04:
you don't need to change this code. 
The pin set in prusa fw is basically a D2, which means your ZMAX pin.
If you use endstops as XMIN, YMIN and ZMIN, you have the ZMAX pin free and you're able to ise it ([https://reprap.org/mediawiki/images/c/ca/Arduinomega1-4connectors.png](url))

boelle, 2018-08-04:
i know....

but still i like to take the extra step and use a pin that is never in use... I have some OCD

maybe we could add function so that we can use one driver for each z stepper 

and if we have one 2130 module for each z stepper we could maybe add dual z endstop so that we dont have to slam the X axis up like prusa does but instead gently just go up and stop when steppers detect that its all the way up

basicly i try to think in to the future and make sure pins are free

but what could be more awesome is to have limited support for full graphics displays... ie just 2 or 4 lines of text on the display would be better than trahsing it and buy another display

cimoalpacino, 2018-08-04:
XMAX is never in use if you don't have max endstops.

If you want to use ZMAX and double endstops, you can still use YMAX, but you'll have to edit some code in FW.

boelle, 2018-08-04:
yep... some code will need to changed but for now i just want to get an overview

and if i use pins on aux 1 and aux2 on ramps i know for sure that nothing will come back later and bite me in the rear end

boelle, 2018-08-04:
right now i'm checking if the sensors for the mk3 is easy to get or to make

and then i have to think if it even makes sense to do all this trouble

thess, 2018-08-04:
@3d-gussner - please lock this topic!

boelle, 2018-08-04:
@thess maybe you could just unsubscribe if you dont want to get emails ?

but then again me and @cimoalpacino might end up making stuff that would benefit the community but since you dont like us having contact we will end up not sharing

3d-gussner, 2018-08-05:
@boelle and @cimoalpacino I agree with @thess that this issue is unrelated to the original post.
So please open another issue and maybe it would make more sense to open it on the Prusa github as you have much bigger community there who could help.