3d-gussner/Prusa-Firmware #43: Filament runout

Agniusm, 2018-01-12:
Is it possible to have NC Y_MAX switch? How to go about changing it when DISABLE_MAX_ENDSTOPS is enabled? I have an idea for filament sensor which requires only 2 wire and it would be great to have it working with normally closed switch (filament out is normally closed)

3d-gussner, 2018-01-13:
@Agniusm Hi saw your post in the forum.
Yes it is possible to have NC FRS switch defined.

I defined the Y_MAX_PIN 24 as 'FIL_RUNOUT_PIN 24' in the pins.h file. As i don't use any other max endstops and to prevent issues i disabled the 'DISABLE_MAX_ENDSTOPS' so there are double assignments to **ONE PIN**!

I am not sure if i understand you correctly:
- You want to make FRS switch by using the E-motor and the E-idler by connecting cables to theses?
- Doing this you have to make sure that the voltage and current aren't too high for the miniRambo ATmega pin, you could destroy your board. I don't know if the motor shaft generates any voltage/current in operation or manual movement. So be careful and analyse that really good. If you have an cheap Arduino board (which you can miss i worst case) you should try that there.

I implemented the FRS for two sensors.
  - One for mechanical switches Signal to GND
  - One for optical switches Signal to VCC
So depending if you plan to set the pin to GND or VCC you should be able to read the status.

As said i am not sure what you are planning to do, I don't own a oscilloscope and proper tools to measure if this could harm the controller board.

Agniusm, 2018-01-13:
Yes, you are correct. I tested it and it works, only in reverse. I dont know how to make it so it works the right way.  When there is no filament, idler touches motor acting as closed switch and in endstop display it shows 1 but it should flicker between 1 and 0 as filament out. 
How to define that? Does it have to do with asigned pin 24?

3d-gussner, 2018-01-13:
Did you try to switch between Fil.RS [S to GND] and Fil.RS [S to VCC] and check if the status stays stable.
You also can connect the printer via Pronterface and use M119 to se the status, just do few M119 fast after each other. Posted also in forum if you could share a picture and the wiring (schematic which cable is connected where)

Agniusm, 2018-01-13:
I havent. Mine is connected to ground. Its simple, no need for schematics.  Essentially you have 2 wires connected to y max on the rambo s signal and ground gnd. Thats all. These wires are shrted on the other end(no filament) or open(filament present). I dont see how vcc and s would do anything ehen nothing is connected to vcc. 
I just need to reverse logic in firmware for that frs.

Agniusm, 2018-01-13:
In the firmware at the moment when filament is present the switch closed.
I need switch open when filament is present, reverse logic.
I dont know how to explain it further as it is so simple. So i need your help to point me to where i need to change what in the firmware

3d-gussner, 2018-01-13:
I think you have a design flaw as the extruder needs to pull out the filament when it runs out of filament. The FRS has to be few cm before. In your case the FRS will trigger as no filament can be griped by the extruder and the rest will stuck and you have to remove it manually.

Agniusm, 2018-01-13:
Is it so complicated to point me to the right direction so i can do all the testing and evaluation and then share if its a viable option? 
All i asked to help me figure out where to reverse logic so if you dont know how or simply dont want to help just say so and i'll be on my way. No need to circle arround something i havent asked. Sorry, dont mean to insult or anything

3d-gussner, 2018-01-13:
Changing between 'S to GND' and 'S to VCC' inverts the logic and dis-/enables the internal pull-up resistor.
If you like to modify the firmware open all files with notepad++ or editor and search for "// FILAMENT_RUNOUT_SENSOR" if i did my job good all modifications should start there and end with "// end FILAMENT_RUNOUT_SENSOR".

In the ultralcd.cpp i use switch function with 3 cases where the 'FIL_RUNOUT_INVERTING', 'fil_runout_status' and 'ENDSTOPPULLUP_FIL-RUNOUT' are set.
In Marlin.cpp just you may need modify the part too.

3d-gussner, 2018-01-13:
No offense taken. Just tried to answer your questions and point out possible issues.
Did you measure if there is any voltage or current generated from the motor?

3d-gussner, 2018-01-13:
By switching between both states it worked for me and others with mechanical or optical switches to get the results stable.

3d-gussner, 2018-01-13:
BTW: the most resent branch is the https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-Private_build

3d-gussner, 2018-01-13:
If you have more question, please feel free to ask. Good luck!

Agniusm, 2018-01-13:
Thanks. That's great.  I tested with multimeter and its 0 volts. When i test continuity it shows up as 940 or close to that, does not beep but it is enough resistance for rambo to work.
Regarding the issue you pointed i was hoping that inertia would feed filament just a bit into ptfe and new filament will push it out the other end. If it works, that means that there is ~0 waste. I had microswitch working with your firmware, although older release for 2 months but i hate adding stuff to the extruder or elsewhere. This would be so elegant i would like to test the concept to the end.
I will report back.

Agniusm, 2018-01-13:
BTW was there any tweaks done to extruder motor movement? somehow my motor stops extruding on few of the prints since fiddling with firmware?
I have changed S to VCC in calibration and it works. Off to test filament loading

Agniusm, 2018-01-13:
So i guess this section needs changing:
`// FILAMENT_RUNOUT_SENSOR
#ifdef FILAMENT_RUNOUT_SUPPORT
static void lcd_fil_runout_status_set() {
	switch (fil_runout_status) {
	case 0:
		fil_runout_status = 1;
		FIL_RUNOUT_INVERTING = 0;
		ENDSTOPPULLUP_FIL_RUNOUT = 0;
		fil_runout_status_set = fil_runout_status;
		FIL_RUNOUT_INVERTING_SET = FIL_RUNOUT_INVERTING;
		ENDSTOPPULLUP_FIL_RUNOUT_SET = ENDSTOPPULLUP_FIL_RUNOUT;
		break;
	case 1:
		fil_runout_status = 2;
		FIL_RUNOUT_INVERTING = 1;
		ENDSTOPPULLUP_FIL_RUNOUT = 1;
		fil_runout_status_set = fil_runout_status;
		FIL_RUNOUT_INVERTING_SET = FIL_RUNOUT_INVERTING;
		ENDSTOPPULLUP_FIL_RUNOUT_SET = ENDSTOPPULLUP_FIL_RUNOUT;
		break;
	case 2:
		fil_runout_status = 0;
		FIL_RUNOUT_INVERTING = 1;
		ENDSTOPPULLUP_FIL_RUNOUT = 1;
		break;
	default: fil_runout_status = 0; break;
	}
  eeprom_update_byte((unsigned char *)EEPROM_FIL_RUNOUT_STATUS, fil_runout_status);
  digipot_init();
  lcd_goto_menu(lcd_settings_menu, 9);
}

static void lcd_fil_runout_status_tune() {
	switch (fil_runout_status) {
	case 0:
		fil_runout_status = fil_runout_status_set;
		FIL_RUNOUT_INVERTING = FIL_RUNOUT_INVERTING_SET;
		ENDSTOPPULLUP_FIL_RUNOUT = ENDSTOPPULLUP_FIL_RUNOUT_SET;
		break;
	case 1:
		fil_runout_status = 0;
		FIL_RUNOUT_INVERTING = 0;
		ENDSTOPPULLUP_FIL_RUNOUT = 0;
		break;
	default: fil_runout_status = 0; break;
	}
  eeprom_update_byte((unsigned char *)EEPROM_FIL_RUNOUT_STATUS, fil_runout_status);
  digipot_init();
  lcd_goto_menu(lcd_tune_menu, 9);
}
#endif
// end FILAMENT_RUNOUT_SENSOR
`
But i have no clue what are those cases and which is responsible for sensor reversing

Agniusm, 2018-01-13:
IS there a mistake in your code in marlin_main.cpp where you have:
`bool fil_funout_inv = true;` 
Probably true when filament runs out fun goes out :D

3d-gussner, 2018-01-14:
Sorry for the late answer. I may need add some comments to make it easier to find.

Easiest for you would be to change under 'case 0' the 'FIL_RUNOUT_INVERTING = 0;' to 'FIL_RUNOUT_INVERTING = 1;' which inverts the logic.

But you also need to change the same thing in the Marlin_main.cpp as with power-up it reads the FRS status and sets the logic and internal pullup-resistor if needed.

Marlin_main.cpp around Line 1285:

// FILAMENT_RUNOUT_SENSOR
#ifdef FILAMENT_RUNOUT_SENSOR
  fil_runout_status = eeprom_read_byte((uint8_t*)EEPROM_FIL_RUNOUT_STATUS);
  if (fil_runout_status == 1) {
	  FIL_RUNOUT_INVERTING = 0;
	  ENDSTOPPULLUP_FIL_RUNOUT = 0;
	  
  } else {
	  FIL_RUNOUT_INVERTING = 1;
	  ENDSTOPPULLUP_FIL_RUNOUT = 1;
  };
  fil_runout_status_set = fil_runout_status;
  FIL_RUNOUT_INVERTING_SET = FIL_RUNOUT_INVERTING;
  ENDSTOPPULLUP_FIL_RUNOUT_SET = ENDSTOPPULLUP_FIL_RUNOUT;
#endif
// end FILAMENT_RUNOUT_SENSOR

You also could add an additional 'case' for NC switches. 

Hope that helps.

3d-gussner, 2018-01-14:
Where did you find? 

'IS there a mistake in your code in marlin_main.cpp where you have:
bool fil_funout_inv = true;
Probably true when filament runs out fun goes out :D'

On my computer and github MK2-private_build branch 'Line 263: bool fil_funout_inv = false;'

Agniusm, 2018-01-14:
Thanks. I will try changes you suggested.
What i meant, should `bool fil_funout_inv = false` be `bool fil_runout_inv = false` You have funout not runout. F is close to R so you might have a typo there.

3d-gussner, 2018-01-14:
If you need more help, please let me know

Agniusm, 2018-01-14:
So in marlin_main.cpp from this:
`FIL_RUNOUT_INVERTING = 0;
	  ENDSTOPPULLUP_FIL_RUNOUT = 0;`
to this:
`FIL_RUNOUT_INVERTING = 1;
	  ENDSTOPPULLUP_FIL_RUNOUT = 1;`
?

Agniusm, 2018-01-14:
That did not work. Perhaps you could look at your end? Could you make so your sensor when filament present shows flickering 0 and 1 in calibration menu and when there is no filament - 1?

3d-gussner, 2018-01-14:
Thanks for the Funout Typo fixed it

3d-gussner, 2018-01-14:
You need to have
FIL_RUNOUT_INVERTING = 0; //setting the logic to NC=0 , NO=1
ENDSTOPPULLUP_FIL_RUNOUT = 1; //activate the internal pull-up resistor as you use GND to signal

in both files.

Marlin_main.cpp
// FILAMENT_RUNOUT_SENSOR
#ifdef FILAMENT_RUNOUT_SENSOR
  fil_runout_status = eeprom_read_byte((uint8_t*)EEPROM_FIL_RUNOUT_STATUS);
  if (fil_runout_status == 1) {
	  FIL_RUNOUT_INVERTING = 0;
	  ENDSTOPPULLUP_FIL_RUNOUT = 0;
	  
  } else {
	  FIL_RUNOUT_INVERTING = **0**;
	  ENDSTOPPULLUP_FIL_RUNOUT = 1;
  };
  fil_runout_status_set = fil_runout_status;
  FIL_RUNOUT_INVERTING_SET = FIL_RUNOUT_INVERTING;
  ENDSTOPPULLUP_FIL_RUNOUT_SET = ENDSTOPPULLUP_FIL_RUNOUT;
#endif
// end FILAMENT_RUNOUT_SENSOR

and in Ultralcd.cpp

// FILAMENT_RUNOUT_SENSOR
#ifdef FILAMENT_RUNOUT_SUPPORT
static void lcd_fil_runout_status_set() {
	switch (fil_runout_status) {
	case 0:
		fil_runout_status = 1;
		FIL_RUNOUT_INVERTING = 0;
		ENDSTOPPULLUP_FIL_RUNOUT = 0;
		fil_runout_status_set = fil_runout_status;
		FIL_RUNOUT_INVERTING_SET = FIL_RUNOUT_INVERTING;
		ENDSTOPPULLUP_FIL_RUNOUT_SET = ENDSTOPPULLUP_FIL_RUNOUT;
		break;
	case 1:
		fil_runout_status = 2;
		FIL_RUNOUT_INVERTING = **0**;
		ENDSTOPPULLUP_FIL_RUNOUT = 1;
		fil_runout_status_set = fil_runout_status;
		FIL_RUNOUT_INVERTING_SET = FIL_RUNOUT_INVERTING;
		ENDSTOPPULLUP_FIL_RUNOUT_SET = ENDSTOPPULLUP_FIL_RUNOUT;
		break;
	case 2:
		fil_runout_status = 0;
		FIL_RUNOUT_INVERTING = 1;
		ENDSTOPPULLUP_FIL_RUNOUT = 1;
		break;
	default: fil_runout_status = 0; break;
	}
  eeprom_update_byte((unsigned char *)EEPROM_FIL_RUNOUT_STATUS, fil_runout_status);
  digipot_init();
  lcd_goto_menu(lcd_settings_menu, 9);
}

static void lcd_fil_runout_status_tune() {
	switch (fil_runout_status) {
	case 0:
		fil_runout_status = fil_runout_status_set;
		FIL_RUNOUT_INVERTING = FIL_RUNOUT_INVERTING_SET;
		ENDSTOPPULLUP_FIL_RUNOUT = ENDSTOPPULLUP_FIL_RUNOUT_SET;
		break;
	case 1:
		fil_runout_status = 0;
		FIL_RUNOUT_INVERTING = 0;
		ENDSTOPPULLUP_FIL_RUNOUT = 0;
		break;
	default: fil_runout_status = 0; break;
	}
  eeprom_update_byte((unsigned char *)EEPROM_FIL_RUNOUT_STATUS, fil_runout_status);
  digipot_init();
  lcd_goto_menu(lcd_tune_menu, 9);
}
#endif
// end FILAMENT_RUNOUT_SENSOR

3d-gussner, 2018-01-14:
The flickering is an indicator that the signal is not 'stable'. You have to make sure that the ATmega 2560 gets valid low or high signals which is done by pull-up or pull-down resistors.

Using signal to GND we can use the internal pull-up resistor function of the ATmega 2560. That's why the S to VCC has ENDSTOPPULLUP_FIL_RUNOUT = 1.

All signal to VCC sensors should have a pull-down resistor to give valid low/high signal. That's why the S to VCC has ENDSTOPPULLUP_FIL_RUNOUT = 0.

Hope that explains a bit and why i don't like to change the status when it is flickering, as you could get false positive readings even there is filament present or not depending on your sensor.

Agniusm, 2018-01-15:
OK. Now it starts printing and after 10 or so seconds does the filament change routine. Should i try and change ENDSTOPPULLUP_FIL_RUNOUT = 1 to 0?

Agniusm, 2018-01-15:
Interesting. When i extrude filament through menu, there is no connection between 2 FRS wires but when it starts printing actual print it shows that there is some resistance?!

3d-gussner, 2018-01-16:
@Agniusm I think that the extruder generates during printing some current which troughs off the FRS readings. The ENDSTOPPULLUP_FIL_RUNOUT should be fine now but see below.

Troubleshooting could be:
- Testing just by using the wires (shortening and open) you should see if the firmware works how you expect to.
- Test without printing (steppers disabled)
- You could try to measure the current between the two wires (not connected to miniRAMbo) while printing, best would be kind of oscilloscope but maybe you can also use a multimeter.

I still think that the FRS needs to be few cm before the extruder motor so in case of runout the extruder motor CAN push out the rest of filament from the hotend.

Hope that helps.

3d-gussner, 2018-01-18:
@Agniusm How is it going? Did you find out something new?

Agniusm, 2018-01-19:
Hi, no, had nora virus so off for a bit :)

3d-gussner, 2018-01-19:
@Agniusm Hope you get well soon.

3d-gussner, 2018-01-26:
@Agniusm Hope you are well again. Did you have time to investigate the issue?

Agniusm, 2018-01-26:
Hi. No, had no time. Had to get other diy printer running. I will be doing zaribo mod on prusa soon and i will look into it again then. 
BTW, i have noticed my extruder running hot to the point where it stops extruding in an enclosure. I reverted back to original firmware and its good again. I dont know if it is coincident or there is something causing this in moded firmware

3d-gussner, 2018-03-21:
@Agniusm How is it going?

Agniusm, 2018-03-21:
Hi. Doing great. Have almost done prusa conversion. Abandoned my initial idea for filament sensor and designed another one with microswitch. It basically houses filament switch, filter and ptfe tube for relieving and evening out filament tension.
[image]
[image]

3d-gussner, 2018-03-21:
Looks good, glad you found a solution. Guess the sensing on the gear had some strange feedback so it didn't work. But now we know it!

3d-gussner, 2018-03-28:
@Agniusm Can i close the github issue?

Agniusm, 2018-03-28:
Hi. Sure.

3d-gussner, 2018-03-28:
@Agniusm Thanks

closed 2018-03-28 by 3d-gussner