3d-gussner/Prusa-Firmware #46: Boot warning message stay forever

sn4k3, 2018-02-19:
I'm having an annoyance with this, message stay forever, printer will not work util i click the knob, also octoprint will not connect util that message gone. For enclosure printers this is very anoying.
Is it possible to put a timeout or make it to show only once after flash?

Note: I turn ON and OFF printer often, only when needed...

3d-gussner, 2018-02-20:
@sn4k3 Sorry for the late answer. Which message do you mean?
If it is "Distance between ..." which is implemented by Prusa you need to run the Wizard OR 'First layer cal.' under Calibration. This will store the information in the EEPROM as soon you did it.

If not can you send me more information?
Which firmware did you install? MK2/s, MK2s MMU or MK3?
What is the exact message on the LCD waiting for acknowledge?

sn4k3, 2018-02-21:
Hello,

The last firmware for MK3 (v3.1.1-HP), it warns about an altered firmware version at every boot blocking octoprint from connecting and slow whole web interface until i click the knob and message gone. For enclosured printers it's a pain.
I'm current printing, when i finish i will reboot printer and then copy the exact message and paste here.

Also i have a question about this, before using your firwamre i sent the a,b,c,d,e,f,g,h compensation via gcode, using your firmware and set it under the calibration menu will have same behaviour? Ignore pinda readings and use the defined values? What happen if i have 0 at a point? Pinda will read that point or still ignore?

Regards

sn4k3, 2018-02-21:
Message is: "WARNING: This is an unofficial, unsupported build. Use at your own risk!"

3d-gussner, 2018-02-21:
Thanks for the update. It looks like Octoprint checks the firmware...have to look into that.
- Gcode a,b,c,d,e,f,g,h is one option which goes to +-200 (stock firmware +-100) the other is the Bed level cal. which goes to +-99 (stock +-50). They both do the same.
If you can level your bed via LCD menu you don't need extra gcode, handy if you have different printers printing the same gcode or you share your gcode with others.
- The PINDA fix is made due to an calculation error that can happen during PINDA Temp calibration. Read the https://github.com/prusa3d/Prusa-Firmware/pull/439
You can use gcode D8 to show the values
- "What happen if i have 0 at a point? Pinda will read that point or still ignore?" Do you mean if you have bed level correct 0? The PINDA will always read the distance and the firmware will calculate the bed level with both values.

3d-gussner, 2018-02-21:
@sn4k3 Which Octoprint plug-ins do you have installed can you share a simple gcode that you try to print via Octoprint?
When i connect to my MK2 with newest Octoprint build and Plugin Firmware updater it works without any problems.

3d-gussner, 2018-02-21:
@sn4k3 Found the warning it in the MK3 firmware...have a look into that

3d-gussner, 2018-02-21:
@sn4k3 Found it. Prusa added a function on 11 January 2018 to notify users about the firmware they are using:
- GOLD: No messages shown
- BETA: Beta message shown and if i am not wrong waiting for user
- ALPHA: Alpha message shown and waiting for user
- DEVEL: Shows 'Development build !!" and waits for user
- DEBUG: Shows 'Debuggin build !!!" and waits for user
- UNKOWN: Shows this warning you are getting.
As 'UNKOWN' is default in their github and everybody who is modifying it to it own needs will get this message until they modify it to 'GOLD' or change the code.

Have to think about how to fix this. I could change it to 'GOLD' but as it is a modified firmware people should be aware of it. Kind of like the Prusa approach even makes it now bit hard for us.
Maybe there is a way to change it to show the message with a time out. Have to think about it.

If you want to compile your OWN flavor feel free to use the source code from my the latest release and modify in the 'configuration.h' file #define FW_DEV_VERSION FW_VERSION_UNKNOWN' to '#define FW_DEV_VERSION FW_VERSION_GOLD'.

As I don't own a MK3 yet never i couldn't test it. BUT people asked for a Hyperfine Bed Leveling MK3 version and i compiled and shared it.
Thanks for reporting. Will add this to the Release notes.

sn4k3, 2018-02-21:
That message will show alone at printer startup, no octoprint needed, but in fact it will block octoprint connection.
When i flashed the original most recent Prusa firmware it show a message telling the PID values were reseted, and that message show only once, like one time notify to the user. Would be cool to do the same with the firmware if possible. I think people who install custom firmwares are aware of it and don't need to be warned every single time :)
As i told very anoying for enclosured printers, flash air or octoprint printers where you turn on and go away, since octoprint reset printer on connection message show twice (On printer start and on octoprint connect).
If printing from SDCard will not be so anoying

Regards

3d-gussner, 2018-03-21:
@sn4k3 Hi, sorry no update yet. Got sick and coding isn't a good idea at this moment. Hope next week i will have FW 3.1.3 (or newer) with Hyperfine bed leveling ready. I am also thinking to add Linear advance 1.5 as experimental version.

3d-gussner, 2018-03-28:
@sn4k3 Hope getting this done this or next week. Feeling fit enough to merge the Hyperfine Bed Leveling and other things.

sn4k3, 2018-03-28:
Nice, i will happy test the new build :)

3d-gussner, 2018-03-30:
@sn4k3 Hi just compiled 3.1.3-HP firmware. It would be great if you could give some feedback before i make it stable release.
The LCD message should be gone BUT as Prusa added few new features needing the same EEPROM space you HAVE to write down your existing hyperfine bed leveling values set them to 0 and THEN flash the firmware 3.1.3-HP. After flashing you can re-add the values to your printer.
Here the link:
https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.3-HP

sn4k3, 2018-03-31:
Sure, i will tune my printer and once that i will flash and recalibrate use your fw. I really think this sould be merged into official, at least the hyperlevel bed...
Thank you

3d-gussner, 2018-03-31:
@sn4k3 Thanks for testing and i hope for a good feedback. Please also check the PINDA temp calibration if it works for you or not.

3d-gussner, 2018-03-31:
@sn4k3 Just found a found a bug. Please go back to stock firmware or the 3.1.0-HP version. Still have to  investigate why D8, M75 and M76 are not working.
For D8 i may have found why. But M75 and M76 just should work.
Keep you updated, and BIG thanks for testing.

Hope getting my MK3 soon to prevent users testing this firmware.

sn4k3, 2018-03-31:
No problem, i still not finish print some parts i need for the tune :) But thanks for warn.
Regards

3d-gussner, 2018-03-31:
Can you try G75 and G76 i guess messed up with M and G 👎
Even i checked the code twice yesterday i messed up to allocate EEPROM space twice, SORRY. 
The D8 issue still don't know why it is not working, but don't give up the hope to find it.

sn4k3, 2018-04-01:
Sure, can you also merge the gcode for read pinda temperature? Its very usefull addition.

3d-gussner, 2018-04-01:
@sn4k3 which gcode read?

sn4k3, 2018-04-01:
you need to implement, check the official github under pull requests

3d-gussner, 2018-04-04:
@sn4k3 which one exactly?
https://github.com/prusa3d/Prusa-Firmware/pull/542
https://github.com/prusa3d/Prusa-Firmware/pull/543

Did you try yourself to get the D codes running? If so what did you try?

3d-gussner, 2018-04-04:
@sn4k3 Hi can you please try https://github.com/3d-gussner/Prusa-Firmware/releases/tag/MK3-v3.1.3-HP?

sn4k3, 2018-04-04:
prusa3d#542

M666 S35 ; the new code - wait until PINDA is >= 35C

3d-gussner, 2018-05-02:
As soon the Prusa firmware looks stable (the 3.2.0-RC1 had way too much issues) i will merge Hyperfine-Bed-Leveling into the firmware.

3d-gussner, 2018-09-07:
Close this issue due no interaction.

closed 2018-09-07 by 3d-gussner