3d-gussner/Prusa-Firmware #47: Please add stahlfabrik PINDA fix

Crunch69, 2018-03-12:
Please add stahlfabrik PINDA fix

Crunch69, 2018-03-12:
https://github.com/stahlfabrik/Prusa-Firmware/tree/MK3_FIX_PINDA_TEMP_COMP
https://github.com/stahlfabrik/Prusa-Firmware/tree/MK3_GCODE_TOOL_FOR_TEMP_COMP_ACCESS
https://github.com/stahlfabrik/Prusa-Firmware/tree/MK3_PINDA_TEMP_WAIT_GCODE

3d-gussner, 2018-03-15:
Thanks for the hint. Have to see what it is about.

3d-gussner, 2018-03-21:
Saw a new a new pull request https://github.com/prusa3d/Prusa-Firmware/pull/552 Lukas investigated the PINDA temp calibration and i think that is the right way to go. Have to as @stahlfabrik what his PM was about.

3d-gussner, 2018-03-28:
@Crunch69 Still waiting for results feedback from Prusa.

3d-gussner, 2018-03-31:
@Crunch69 Have a new pre-release 3.1.3-HP hope getting feedback soon so can pubish it.
Still don't get why the PR form lukas didn't make it he wrote great paper about the PINDA temp calibration.

3d-gussner, 2018-04-07:
@Crunch69 Can you PM me via twitter or facebook?

3d-gussner, 2018-05-02:
@Crunch69 Sorry for the late answer, but as there lot of issues with 3.2.0-RC1 i stopped to merge things and will wait until Prusa has kind of stable version.

stahlfabrik, 2018-05-02:
Never mind my fix! Just make sure to be based on 3.2.0-RC2 which has my two new gcodes (M860, M861) to setup the PINDA calibration values AND also supports NEGATIVE offsets - thus making the need to change the sign of the mathematical operation unnecessary. (For some/ a lot of printers the original sign might be correct - for mine it was not)