3d-gussner/Prusa-Firmware #49: MK42 heatbed on Einsy board with 12v e3d and filament sensor and Pinda 2 probe

darkdelphi, 2018-03-20:
Ok so what I am looking for is how to change the Mk3 firmware to accomadate a 12v power supply and the mk42 heatbed but make use of the original pinda 2 probe and the prusa filament sensor. 
I have an original Einsy board from Ultimachine and an original E3D V6 12v. I also got the original bondtech   gears and have completely assembled a clone hotend (made certain to use black filament) 

Goal is eventually upgrade to full MK3 once the spring steel sheets are available for immediate purchase. I don't preorder as a rule.

darkdelphi, 2018-03-21:
The fans I didn't even consider so that you very much, I just got those ordered and will have them tomorrow. If I were to go ahead and use 24v power supply where would could I still use the 12v extruder and bed, if it is simply a matter of changing the PID settings where are those located. or would it be simpler to turn on the einsy features on the mk2.5 12v variant of the firmware.

3d-gussner, 2018-03-21:
@darkdelphi You should not run a 12V heater cartridge or 12V MK42 bed on 24V, they can (probably will) fail.

- Changing the PID you should do on all printers as every is bit different, BUT it will not help you if you go for 24V PSU with 12V heater cartridge and 12V heated bed.

- Changing the heater cartridge to an 24V isn't a problem https://e3d-online.com/heater-cartidge-40w or you could ask prusa to get one.

- The MK42 bed is 12V and could be used with a 24V to 12V step down converter, but this would need a big one which could handle the needed current. At the end a MK52 bed could be cheaper.

As you have most parts now adding the 24V 40W heater cartidge + 24V MK52 bed + spring sheet and a 24V PSU this may be cost inefficient. It had been 'cheaper' to get the MK3 Upgrade kit.

The MK25 firmware is made for the miniRAMBO with an MK52 12V bed and not for the EINSY with MK42 bed. You could get some settings from that.
At the end you gonna need your own firmware with EINSY 12V + 12V E3dv6 + MK42 12V bed.

Sorry, it isn't that easy.

I ordered the MK3 Upgrade kit, 'GT-16 pulleys' and '623h Bearing with housing' from Prusa. Get some other parts like motors, rods, bearings, LCD, GT2 belt and fasteners from other sources and build a custom MK3 printer.
But my firmware changes will be much smaller than yours and so easier to maintain.

I love my MK2 with the miniRambo and think it has too much value in parts to switch everything, so i change few things on that one:
- Got a 12V noctua fan from Amazon and will connect it to the miniRAMbo (it is supported in the MK25 firmware
- Ordered the PINDAv2 which will also be connected to the miniRAMbo (MK25 firmware)
- Should get a spare bondtech gear out of the ordered MMU kit (MK25 firmware) that may change as the MMU v2.0 is coming and i have to order one extra.
- Should get a spare filament sensor out of the MMU upgrade kit MK2/MK25 to MK3 (MK25 firmware)
that may change as the MMU v2.0 is coming and i have to order one extra.
- I can use the MK2 firmware code settings for the MK42 bed i am using.
- As soon the powder coated spring steel sheets are available i will order it and use it on my MK2.3 or as spare on the custom MK3.

My changes are very close to the Prusa firmware releases so the custom changes will be minimal:
- MK25 firmware + custom height + MK42 bed settings for my MK2.3
- MK3 firmware + custom height for my custom MK3

3d-gussner, 2018-03-21:
Also to get your printer as close as possible to the MK3 you gonna need a 12V power panic circuit, which isn't available to buy even in 24V. You can get the https://shop.prusa3d.com/en/mk3-printer-parts/207-mk3-PSU-24v-240w.html but then you are close to MK3 upgrade kit costs and need a 24V MK52 bed + spring steel sheet + 24V heater cartridge, etc.

Here more info about the https://github.com/prusa3d/MK3_Power_Panic

3d-gussner, 2018-03-28:
@darkdelphi How is it going with your project?

darkdelphi, 2018-03-30:
Oh I was able to get the 5v Noctua, a 24v e3dv6 , as well as an original  Prusa MK3 thermister as well as an appropriate 24v 240w power supply,

3d-gussner, 2018-03-30:
Sounds great. Good luck with your project

3d-gussner, 2018-04-01:
Did you read this ? https://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/einsy-rambo-on-mk2--t6137-s30.html#p74913

3d-gussner, 2018-05-02:
Hi @darkdelphi how is it going? Do you think this issue should still be open? If not please close it.

3d-gussner, 2018-09-07:
Close issue due no active interaction for long time.

closed 2018-09-07 by 3d-gussner