3d-gussner/Prusa-Firmware #53: Values reported on Serial incorrect

madhunm, 2018-07-09:
Hello @3d-gussner 

I have an issue with the 3.1.0 and 3.1.0-001 versions of the firmware when uploaded onto a MK2. 

It is as below:

The temperature values reported on the LCD are correct. However, the values reported over the serial are an order of magnitude off.

For example, with the hotend turned off, I am showing the temperature of the hotend at 30C. The M105 command reports the below values: T311

The same is true for a G81 command; here is a sample output:

`Measured points:
  063833  084815  100204  110000  114204  112815  105833
  056713  076814  091758  101546  106178  105653  099972
  049241  069348  084848  095741  102027  103706  100778
  041417  062417  079472  092583  101750  106972  108250
  033241  056021  075632  092074  105348  115453  122389
  024713  050159  073326  094213  112820  129147  143194
  015833  044833  072556  099000  124167  148056  170667`

I would really like to use the HFM feature; please help!

3d-gussner, 2018-09-07:
@madhunm So so sorry that i didn't answer on your issue. Guess didn't get a notification and as i don't have that much issue reports i don't look that often.

So is that issue still there or is it solved?
If it is still there which serial communication program do you use?
I have no issues with Pronterface, Slic3rPE, OctoPrint and ESP3D to communicate with this firmware version (hex file) on my MK2.

3d-gussner, 2018-09-13:
@madhunm Any update

3d-gussner, 2018-09-24:
Close issue due no update

closed 2018-09-24 by 3d-gussner