3d-gussner/Prusa-Firmware #54: Prusa Mk2s

strongbow1, 2018-09-06:
After bed levelling, it goes to start then runs to the right spits out the filament asking to load filament. I load new Filament then the cycle repeats itself all the time.

3d-gussner, 2018-09-06:
Hi,  did you read the release notes carefully? As this firmware expects a filament run out sensor connected, you need to to the LCD menu disable/enable/disable the FRS if you don't connect a sensor.

strongbow1, 2018-09-06:
Yes the FRS is disabled 

Mit Freundlichen Grüsse
With Best Regards
Sandro Strimer

3d-gussner, 2018-09-06:
Moin moin.

Some users have that issue some not, so please disable/enable/disable it to make sure that the right value is written to the eeprom.

3d-gussner, 2018-09-07:
Moin moin,

did it work?

3d-gussner, 2018-09-07:
Great,  thanks for the feedback. Glad you are building a FRS for your MK2/s :-)
Read the FRS section in the release notes. You can use M119 to see the status (if enabled) and also turn it on/off (no config change) during the print via tune menu.

Would be great if you could post a picture of your FRS and how Hyperfine Bed Leveling improved your printer.

3d-gussner, 2018-09-13:
How is it working?

strongbow1, 2018-09-13:
Is going good. Opted for optical sensor. Am waiting for the parts to arrive.

Mit Freundlichen Grüsse
With Best Regards
Sandro Strimer

3d-gussner, 2018-11-05:
Close the issue ... thanks for the feedback

closed 2018-11-05 by 3d-gussner