3d-gussner/Prusa-Firmware #58: Forum links dead

Moonchill, 2019-05-16:
Prusa seems to have switched to a new forum setup, which means all of the forum links in the article are no longer working. Some of the threads you can find again by searching the forums, but that doesn't really work for very long threads where the link is to a specific post many pages into the thread.

3d-gussner, 2019-05-21:
@Moonchill Thanks, i will try to update these in the release notes. Or maybe even copy important things.
Sorry for the late answer.

3d-gussner, 2019-05-21:
Just updated the release notes:
https://github.com/3d-gussner/Prusa-Firmware/releases/tag/MK25-v3.1.3-HP
https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM
and Wiki
https://github.com/3d-gussner/Prusa-Firmware/wiki/How-to:-Use-Hyperfine-Bed-Leveling

Hope didn't  forget one

Moonchill, 2019-06-05:
This got lost in my mail box. Links are fixed, and again sorry for the slow response. Closing this.

closed 2019-06-05 by Moonchill