3d-gussner/Prusa-Firmware #60: Question: How to add ramps board

boelle, 2019-09-25:
Sorry for the abuse of the issue list, but i want to try and port the prusa firmware to run on ramps/mega

what would i need to change? just asking for pointers so i have a starting point, 

i have a mendel90 printer that i almost run as a mk2.5, i dont think that using the mk3 would be wise as that would require to much hardware mods to the ramps board

will close this one again so it does not clutter the issue list to much

closed 2019-09-25 by boelle

boelle, 2019-09-25:
trying this one: https://www.thingiverse.com/thing:2607463

i just had to figure the difference between mk2 and mk2.5 and what the S brings to the plate

cimoalpacino, 2019-09-26:
I've ported Prusa fw version 3.6.0, 3.7.0 and 3.8.0 to ramps.
https://github.com/cimoalpacino/Prusa-Firmware/tree/MK3_3.8.0_RAMPS?files=1

Ps: version 3.8.0 v2 from my github does not work after i tested it, due to 16-bit timers used. V1 should work fine

boelle, 2019-09-26:
@cimoalpacino will have a look at it, but is it for the mk2 or mk3 ?

boelle, 2019-09-26:
i looked at the files and i see ramps added, but not in the files in the /Firmware/variants/ folder

so are there a bit of setup to do that i have missed?

boelle, 2019-09-26:
@cimoalpacino think i got it

you took the config options from the files in 

and put them in Firmware/Configuration.h ?? i see a lot of /*RAMPS*/

3d-gussner, 2019-09-26:
Just a question: Why don't you fork it yourself and add the Issue tab?

3d-gussner, 2019-09-26:
Don't mis understand me wrong.  It is okay for me you abuse this here, but to document your issues it would make more sense if you have it in your own repository.

boelle, 2019-09-26:
hehe, its more me asking questions to understand what he did. its not an issue as such, just my head that is a "tad" slow :-D

3d-gussner, 2019-09-26:
@cimoalpacino you can go under settings in your fork and activate issue reports. That would make more sense, than having the discussion somewhere else than here.
@boelle NP I was just surprised to get an email notification that there was a Issue report in my github. As it is quite time ago that I released any firmware.

3d-gussner, 2019-09-26:
@boelle he has issue open in his https://github.com/cimoalpacino/My-Prusa-Firmware/issues repository

boelle, 2019-09-26:
and i just posted there :-D just a few questions to get my head going

3d-gussner, 2019-09-26:
I like the idea and my have a look at @cimoalpacino fork ...

boelle, 2019-09-26:
yep, the reason why i'm after this is that i have a frankenbuild of a mendel90 with prusa x axis and mk42 bed (with pinda probe of course)

i use prusa slicer and i have selected the mk2s printer and only changed the start and stop gcode and i added octoprint settings, other than that i run my printer as a mk2s

and since marlin keeps changing stuff i dont like i got the idea: hey you have a mega2560 and you got those prem. ramps boards from spain, why not see if you can run prusa firmware on it

cimoalpacino, 2019-09-26:
sorry to spam @3d-gussner :)  
will reply to @boelle back on my page.
cheers

boelle, 2019-09-26:
and if it helps blame me for the spamming, i did start it :-D
(also the reason i closed it so quick)