3d-gussner/Prusa-Firmware #61: Turkish language support for Zaribo Firmware

fsadak, 2019-12-04:
Turkish language files.
[lang_tr.zip](https://github.com/3d-gussner/Prusa-Firmware/files/3922965/lang_tr.zip)

3d-gussner, 2019-12-04:
@fsadak Thanks.  Is there any rule how äöü... and other can be written differently?
see https://github.com/3d-gussner/Prusa-Firmware/blob/MK3/lang/lang-import.sh

In German you are allowed to write "ä" as "ae", "ö" as "oe", "ü" as "ue" and "ß" as "ss".
Or in French "é" as "e", "è" as "e".

Would be great if we also could add these rules to this script. So if somebody in the future forgets not to use "üöä" the script will replace them.

fsadak, 2019-12-04:
Yes we can set some rules about language specific letters. Here is the list;
Lower cases:
"ı" as "i", "ğ" as "g", "ü" as "u", "ş" as "s", "ö" as "o", "ç" as "c"

Upper cases:
"İ" as "I", "Ğ" as "G", "Ü" as "U", "Ş" as "S", "Ö" as "O", "Ç" as "C"

Best regards...

closed 2023-08-24 by github-actions[bot]