3d-gussner/Prusa-Firmware #66: [ENHANCEMENT] Add MAC OS support to PF-build.sh

3d-gussner, 2019-12-18:
Enter what type of printer or upgrade the enhancement applies to.
MK3S, MK3, MK2.5S, MK2.5
MMU2S, MMU2

**Is your enhancement related to a problem? Please describe.**
As I don't won a MAC it would be great if some could help me to get `PF-build.sh` also on MAC.
At this moment supported OS are
- Windows 10
   - git-bash
   - Windows Linux Subsystem (Ubuntu and Debian have been tested)

**Describe the solution you'd like**
Add MAC OS to work with PF-build.sh.

3d-gussner, 2019-12-18:
Hello @bgiot,

I have seen that you have a Prusa Firmware fork and created your own build environment for MAC OS.
Question: Can you help me to get `PF-build.sh` running on MAC OS?

bgiot, 2019-12-21:
I used platform.io, not arduino...

closed 2020-02-04 by 3d-gussner