3d-gussner/Prusa-Firmware #68: 4x6 Mesh Bed Leveling

gregsaun, 2020-02-12:
**Printer type** - Bear 2.1 MK3S with Hemera extruder
**MMU Upgrade** - None

**Is your enhancement related to a problem? Please describe.**
With the increase of different type of extruders on Prusa printers, it would be interesting to have an easy mesh bed leveling that naturally avoid magnets
  
**Describe the solution you'd like**
The solution would be a 4x6 MBL as seen in this picture:
[image]

Please find attached Fusion 360 files for 24v and 12v MK54 beds.
[4x6_mesh_bed_leveling.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4194588/4x6_mesh_bed_leveling.zip)

Moorviper, 2023-08-22:
I tried that grid on my bear 2.1 with rrf and it looks like that it works relatively good
[image]
[image]

Moorviper, 2023-08-22:
I did the nylock mod and this is how the mesh looks like

[image]
[image]

closed 2023-10-30 by github-actions[bot]