3d-gussner/Prusa-Firmware #69: [Question] What files were changed to enable OLED support.

devsfan1830, 2020-04-08:
I have a Haribo 3030 modded MK2.5S. So basically a stock printer with a slightly larger Z. I found a link to this firmware that stated you added OLED support for the Winstar WEH002004AWPP5N00000. 'd love to get mine working again as the modded files I used dont work on the 2.5S firmware. They seemed to have overhauled the LCD stuff. I'd like to know if the changes you made can be back ported to MK2.5S and id so, what changes I need to make. I'm familiar with how to compile firmware since I need to do it to fix my Z max with each new stock firmware. Any help you could provide would be awesome. Thanks!

3d-gussner, 2020-04-11:
@devsfan1830 Sorry for the late answer.
You can use https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_WEH002004_OLED or compare that branch to the Prusa MK3 branch to see what have been changed.

Hope that helps.

3d-gussner, 2020-04-11:
also have a look at https://github.com/vertigo235/Prusa-Firmware/tree/MK3_3.9.0/ he has some awesome custom variants as well

devsfan1830, 2020-04-11:
Comparing to the current state of the prusa branch it looks like your modification is identical aside from the OLE specific code. So i'm going to try a compiled version of your branch which generated a 3.9.0-RC1. The further modifications to my printer for the bear extruder and 225 Z height dont impact the OLED code so it SHOUKD be fine aside from any bugs prusa still has in the RC1. Thanks for the help!

closed 2020-04-11 by devsfan1830