3d-gussner/Prusa-Firmware #71: 📑Report: Seam test results

3d-gussner, 2020-05-03:
Please report here your Seam test results and please report also back GOOD results not only issues.

- Printer: MK2.5, MK2.5s, MK3, MK3s
- MMU: MMU1, MMU2, MMU2s
- Firmware version printer and MMU
- Bad to good scale : 0 - 10

Feel free to write additional information you want to share.

I will try to update following table so everyone can see what results we got from the community.

|Printer | MMU | PFW | MFW | Nearest | Random | Aligned | Rear |
| ------ | ------- | ---------- | ----------- | --------- | --------- | --------- | ------- | 
| MK3s | MMU2 | 3.9.0-RC2 | 1.0.6 | 8 | 7 | 8 | 7 |

3d-gussner, 2020-05-03:
Example report form user:
I tested on Mk3s 3.8.1 no mmu with K45 Nearest=8 Rand=7 Aligned=8 Rear=7
3.9.0-RC3 with K45 Nearest=8 Rand=7 Aligned=8 Rear=7
3.9.0-RC3 with K0.2 Nearest=8 Rand=7 Aligned=8 Rear=7

Vossberger, 2020-05-05:
I tested MK3s no MMU
with FilamentumPM (PLA Orange)
3.8.1        with K33   Nearest=7  Rand=7  Aligned=7  Rear=7
3.9.0-RC3 with K33   Nearest=8  Rand=7  Aligned=8  Rear=7
3.9.0-RC3 with K0.07 Nearest=8  Rand=8  Aligned=8  Rear=7

mmuellerphoto, 2020-05-06:
I tested Prusa Mk3s, no MMU, Redline Filament PLA Silver, Ruby Nozzle, SD-Card
3.8.1 with K41 Nearest=6 Rand=5 Aligned=6 Rear=6
3.9.0-RC3 with K41 Nearest=7 Rand=6 Aligned=6 Rear=6
3.9.0-RC3 with K0.14 Nearest=7 Rand=6 Aligned=7 Rear=7

Attached an image of the 3.9.0-RC3 with K0.14 Test ... 
[image]

closed 2023-08-24 by github-actions[bot]