3d-gussner/Prusa-Firmware #72: 📑Report: Retraction test results

3d-gussner, 2020-05-06:
Please report here your Retraction test results.

Use the 3mf file below and try to pint it with PETG or some other material that tends to be stringy.
[Container_50x20_NoCore_v2.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4585431/Container_50x20_NoCore_v2.zip)

- Printer: MK2.5, MK2.5s, MK3, MK3s
- MMU: MMU1, MMU2, MMU2s
- Firmware version printer and MMU
- Bad to good scale : 0 - 10

Feel free to write additional information you want to share.

I will try to update following table so everyone can see what results we got from the community.

|Printer | MMU | PFW | MFW |K value| Nearest | Random | Aligned | Rear | User |
| ------ | ------- | ---------- | ---- | ------- | --------- | --------- | --------- | ------- | -------- |
| MK3s | MMU2 | 3.8.1 | 1.0.6 |45| 7 | 7 | 7 | 7 | Average |
| MK3s | MMU2 | 3.9.0-RC3 | 1.0.6 |45| 8 | 7 | 8 | 7 | Average |
| MK3s | MMU2 | 3.9.0-RC3 | 1.0.6 |0.2| 8 | 8 | 8 | 7 | Average |
|  |  | |  |  |  |  |  |  |  |

Vossberger, 2020-05-07:
MK3s, no mmu, Ruby Nozzle, Nunus "Transparent hellblau" PETG 

FW3.8.1 K66           very stringy (!), Extrusion up to 160% necessary, Result =2
FW3.9.0-rc3 K66     looks like one complete body, extrusion=100%, Result=0
FW3.9.0-rc3 K0.11  no string, extrusion=100%, Result=9

(Photo from bottom to top : FW3.8.1 / FW3.9.0rc3 LA1.0 / FW3.9.0rc3 LA1.5)
[image]

wavexx, 2020-05-07:
Who-ho.. massive difference. I'm missing part of the discussion. How K66 /  K0.11 are chosen?

Vossberger, 2020-05-07:
> Who-ho.. massive difference. I'm missing part of the discussion. How K66 / K0.11 are chosen?

I used https://marlinfw.org/tools/lin_advance/k-factor.html and set the K-Value best for this filament for FW3.8.1 and did a second round for FW3.9-rc3 with LA1.5

3d-gussner, 2020-05-08:
The LA1.0 to LA1.5 conversion is way off for this filament with K=0.21 vs. real LA15 K=0.11.
So the issue is in the conversion OR in the initial LA1.0 K value
```
Send: M900 K66
Recv: LA10C: Linear Advance mode: 1.0
Recv: LA10C: Adjusted E-Jerk: 4.50
Recv: echo:Advance K=0.21
Recv: ok
```
**Looking at other PETG profiles**, they all have LA1.0 M900 K45 which is very close to the LA1.5 K=0.11
```
[...]
Send: M900 K45
Recv: LA10C: Linear Advance mode: 1.0
Recv: LA10C: Adjusted E-Jerk: 4.50
Recv: echo:Advance K=0.13
Recv: ok
```
@Vossberger Can you please rerun the k-factor for **LA10** for your PETG? I really think there is the problem as all other tests/results with PETG were nearly the same as with FW3.8.1

Vossberger, 2020-05-25:
Once again to this test set. It is possible that I did not adjust the eSteps before changing the FW again (I use a 0.9 stepper [415 -> 830].

closed 2023-08-24 by github-actions[bot]