3d-gussner/Prusa-Firmware #73: 📑Report: Speed test results

3d-gussner, 2020-05-07:
Please report here your Speed test results.
Best to print it with FW3.9.0-RC3 or newer with **LA1.5 K values.**
Feel free to print with FW3.8.1 and LA1.0 K values.
Play around with the speeds and report also the sweet spot speed you think deliver a "good ++" print

Use the 3mf files below.
Look at the wall artifacts, corners (sharp vs round), ghosting text, general finish

[Speed_corner_arc_test_200x100x10_no_top_bottom_infill.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4592709/Speed_corner_arc_test_200x100x10_no_top_bottom_infill.zip)

[Speed_corner_arc_test_200x100x10_no_top_bottom.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4592711/Speed_corner_arc_test_200x100x10_no_top_bottom.zip)

[Speed_corner_arc_test_200x100x10.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4592712/Speed_corner_arc_test_200x100x10.zip)

- Printer: MK2.5, MK2.5s, MK3, MK3s
- Firmware version printer
- Bad to good scale : 0 - 10

Feel free to write additional information you want to share.

I will try to update following table so everyone can see what results we got from the community.

|Printer | PFW | K value | Artifacts| Corners | Ghosting | Quality | max. Speed | User |
| ------ | ---------- | ------- | --------- | -------- | ------- | --------- | -------- | -------- |
| MK3s | 3.8.1 | kk | N/A | N/A | N/A | N/A | N/A | Average |
| MK3s | 3.9.0-RC3 |  k.kk | N/A| N/A | N/A  | N/A | N/A | Average |
|  |  | |  |  |  |  |  |  |  |

mmuellerphoto, 2020-05-08:
Prusa i3 Mk3s, 3.9.0-RC3, 0.13K, "Stock Version" except Ruby nozzle
@150mm/s (via Octoprint): Artifacts 4, Corners 4-6 (depending on corner), Ghosting 3, Quality 3
@125mm/s (via Octoprint): Artifacts 5, Corners 5-7 (depending on corner), Ghosting 4, Quality 4
@100mm/s (via Octoprint): Artifacts 6, Corners 5-7 (depending on corner), Ghosting 6, Quality 5
@100mm/s (via SD-Card): Artifacts 6, Corners 5-7 (depending on corner), Ghosting 6, Quality 5

After one corner in front of test in the object the surface didn't even get closed ... Even not with 100mm/s.

Some pictures of the results:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wavexx, 2020-05-10:
The interesting tidbit here seems that the line is closed where the "Y-axis" is not there.

wavexx, 2020-05-10:
@mmuellerphoto could you just make one try with the 3.8.1, at a fast speed like 150 or 125mms? I'm curious if this got worse. Also, pay attention if the external perimeter is changing direction between the bottom/top area (where there's no writing) and the middle. I'd like to know if that corner is always an accelerating or decelerating, or if it's switching between

Vossberger, 2020-05-11:
=> 3.9.0-RC3, 0.11K, FilamentumPM PLA, SD-Card, 2 x Tests (w/o Infill+Top+Bottom)
@150mm/s : Artifacts 4, Corners 5-6 (depending on corner), Ghosting 3, Quality 7
Wall thikness depending on direction
[Test #73.pdf](https://github.com/3d-gussner/Prusa-Firmware/files/4608177/Test.73.pdf)

mmuellerphoto, 2020-05-11:
@wavexx ... So like promised the additional test ...
Prusa i3 Mk3s, 3.8.1, 44K, "Stock Version" except Ruby nozzle
@150mm/s (via Octoprint): Artifacts 6, Corners 5-7 (depending on corner), Ghosting 5, Quality 5
@125mm/s (via Octoprint): Artifacts 6, Corners 5-7 (depending on corner), Ghosting 6, Quality 6

No holes anymore - so really surprising result! I used the same files so same GCode like in the first test! Also the thickness is now much better than in the first test. Maximum of 0.05mm off.

[image]

[image]

[image]

[image]

[image]

So I thought: there is something wrong with my K value. Wanted to print 2 Linear advance calibration patterns - one with 3.8.1 and one with 3.9.0-RC3 ...

3.8.1 result - K0 to 80 in Stepping 2:
[image]
So my K44 isn't that bad!?

3.9.0-RC3 result - same file!
[image]

Also a little video "what happened" ...
[IMG_3690.mov.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4611072/IMG_3690.mov.zip)

Shouldn't the Firmware take LA1.0 values and convert it to LA1.5 values?

wavexx, 2020-05-11:
The automatic switching doesn't work out-of-the-box for K patterns of old LA 1.0 test files depending on the values (the first K value generally switches the print in LA1.5 mode directly). If you put a dummy ``M900 K10`` in the start gcode it works.

wavexx, 2020-05-11:
> @wavexx ... So like promised the additional test ...
> Prusa i3 Mk3s, 3.8.1, 44K, "Stock Version" except Ruby nozzle
> @150mm/s (via Octoprint): Artifacts 6, Corners 5-7 (depending on corner), Ghosting [...]

Confused about the same gcode? I guess it's the same model, but with K set to 44?

I'm pretty sure there's a direction change when the "Y-axis" is written. Did you pay attention if the left corner on the "Y" is the first corner immediately preceding "Y-axis", or the last just _after_ "Y-axis"?

There's some extra-bulge close to the "Y", which looks like it's exactly what's missing on the corner looking at the picture.

mmuellerphoto, 2020-05-11:
> The automatic switching doesn't work out-of-the-box for K patterns of old LA 1.0 test files depending on the values (the first K value generally switches the print in LA1.5 mode directly). If you put a dummy `M900 K10` in the start gcode it works.

Thanks - will give it another try tomorrow.

> Confused about the same gcode? I guess it's the same model, but with K set to 44?

Due to my information it's possible to have both LA values in the GCODE - wrong value will be skipped? In my case it's:
`...
M900 K0.13 ; Filament LA1.5 gcode
M900 K44 ; Filament LA1.0 gcode
...`

> I'm pretty sure there's a direction change when the "Y-axis" is written. Did you pay attention if the left corner on the "Y" is the first corner immediately preceding "Y-axis", or the last just [...]

To be honest: I didn't get what to look out for, so I made a video while printing - hope that helps!?! ... Since it's too large here a link to my Dropbox: https://www.dropbox.com/s/b2mgrbptkrvgzjz/issue73_additional_video.mov?dl=0

wavexx, 2020-05-12:
> Due to my information it's possible to have both LA values in the GCODE - wrong value will be skipped? In my case it's:
> `M900 K0.13 ; Filament LA1.5 gcode`
> `M900 K44 ; Filament LA1.0 gcode`

Ah yes! Absolutely correct

> To be honest: I didn't get what to look out for, so I made a video while printing - hope that helps!?!

It does, thanks. It was just to confirm the direction of extrusion while printing!

closed 2023-08-24 by github-actions[bot]