3d-gussner/Prusa-Firmware #76: 📑Report: Octoprint M112 fix

3d-gussner, 2020-05-28:
Please  report back your experience with FW3.9.1 M112 Fix.
Test procedure:
- Begin with actual stable firmware FW3.9.0
- Check if possible with Octorprint 3.12 and Octoprint 1.4.x

What are we looking for?
Issue is that sometime Octoprint sends M112 emergency code which will stop the print.

brinkeb, 2020-05-31:
Did not have the time to do all tests but I´ve just finished a 15 hour print with 3.9.1 dev and Octoprint 1.4.0 without any issues.

Vossberger, 2020-06-04:
Checked with FW3.9.0 Octoprint v1.4.0 "Detailed LCD Progress Plugin V0.2.5" on two printers (Z320 BE) 
=> 5 x Logs related to prints between 2 hours up to 14 hours checked , 
=> NO(zero) M112

second round with FW3.9.1 to come

leptun, 2020-06-04:
@Vossberger the M112 happens because of manual reset of the printer. It should never happen during normal printing.

Vossberger, 2020-06-04:
I checked the serial log file .. as far as I understood it should it should being shown there ? correct or not ?

closed 2023-08-24 by github-actions[bot]