3d-gussner/Prusa-Firmware #80: [BUG] PF-build.sh ": No such file or directory"

WarrenSchultz, 2020-11-23:
**Describe the bug**
Using  WSL2 with Ubuntu, I'm getting ": No such file or directory" when I try to run the PF-build.sh script on 3.91 firmware
  
**To Reproduce**
Pulled source via Github desktop
Followed the steps in the script to prep the environment
Restarted shell
Ran ./PF-build.sh

I'm sure I'm missing something obvious, but I haven't turned anything up in searches that helps.
Thanks!

mesakomarevich, 2020-12-17:
This happens when PF-build.sh is encoded with Windows line endings (CRLF) rather than the Unix standard LF. A simple fix is to open the file in VSCode and change the file encoding type from CRLF to LF, that fixed it for me, as I was having the exact same issue.

WarrenSchultz, 2020-12-23:
That was the trick, thank you!

closed 2020-12-23 by WarrenSchultz

3d-gussner, 2020-12-23:
Sorry for the late answer, must have missed it in the tons of github notifications.
You can install 'sudo apt install dos2unix' to do the same.

Merry Christmas and a happy new year